rkl099/Appinventor-SerialOTG #1: Teensy Serial

daniele-v, 2020-12-25:
Can you insert the support for the serial driver of Teensy boards https://www.pjrc.com/
Vendor ID: 16C0
Product ID: 048A
Thanks

rkl099, 2020-12-25:
This is a part of the PhysicaloidLibrary code that I have not touch yet. Teensy seems to be a cdc_acm device. Can you verify this?
From what I can see the code only test for Vid (with some exception). If not detected then cdc_acm is the default, and the library tries to open the device so it should work with all cdc_acm devices. I have only an Arduino Uno and ProMicro (Leonardo) to test this so I can’t verify it!
Note: my ProMicro demands Rts=1 (Or Dtr=1) to send data. Maybe this should be the default value in the extension. Please try this!

GramJr, 2021-02-02:
Hello! Thanks for great extension!!! I have an OpenCR board from robotis. It is CDC device, and works fine with OTG Serial Terminal app on Android device, but with with your extension serial is not initialized. Maybe it is possible to change manually type of driver? Or maybe you can help another way with this trouble.

rkl099, 2021-02-02:
I don’t have a OpenCR board from robotis so I can’t test this. 
By initialize i guess that you mean open. 
If you succeed to open the device then you can use DriverName() to see what driver is chosen (if any).
Please try what I mention above about the use of Rts/Dtr, but it will only work if you can open the device. 
Note: I you use AI Companion and make some changes with a device open, then you may have to restart AI companion. This sometimes happens even if the device is closed. Once I had to clear memory in Android Settings for AI companion ,to make it work again.

GramJr, 2021-02-02:
I’ve also got Arduino Nano with CH340 - no problems with opening port. But Drivername() for CH340 gives “None”. Doesn’t  initialized = doesn’t opened with Open CR.

rkl099, 2021-02-02:
If I test with my adapters and Arduinos:

Serial_OTG.Initialize()   I do this in block "when.Screen1.Initilize"   This gives you an instance of the driver library.

Serial_OTG.Open() Try to identify the device and init the driver (May have to be called twice to allow access of USB)

Serial_OTG.DriverName() returns a string  ”com.physicaloid_ai.lib.usb.driver.UartWinCH34x” for CH340 and ”com.physicaloid_ai.lib.usb.driver.UartCdcAcm” for Cdc device

Also some USB Cdc devices must be initialized with serial communication before you call open(), to be recognized as a serial device.

This is all I can test for the moment.

rkl099, 2021-02-02:
I got a Feather M0 that is an CDC ACM device that was not recognized by the driver. Now I have examined the Library code and included the M0 Vendor ID and it know works OK. If you can find out the Vid for your board then I can include it in the driver.

GramJr, 2021-02-03:
Good news, no. 1 - VID 0x0483, PID 0x5740.

rkl099, 2021-02-03:
Try this debug version at  rkl099/Appinventor-SerialOTG_Debug 
I will include it in the main extension if OK.
For  daniele-v  above: I ave also included Teensy Vid 16C0. Please try it and report to me!

GramJr, 2021-02-03:
Tested with VID 16C0 (Teensy 4.0) and with 0483 (OpenCR) - works fine!

closed 2021-03-01 by rkl099