rkl099/Appinventor-SerialOTG #2: Support for FTDI 4232HL (multiple ports)

agregg18, 2021-01-04:
Hi, first thank you for this addition to AI! I can verify that it works great on my FTDI 232RL, but I can't get it working on my 4232HL chip, which is a 4-port chip. When this chip connects to Android, it creates 4 ports (eg. 0-3). I believe your extension is only set up to use Port 0. If I use the FTDI terminal app (https://play.google.com/store/apps/details?id=com.ftdi.j2xx.hyperterm), I can select a port other than 0. Would it be possible to add FTDI Port selection in your extension?

rkl099, 2021-01-07:
Does it work for port 0?
I have no 4232HL so I can't test this.

agregg18, 2021-01-09:
It works great for Port 0 on the 232RL. Unfortunately my 4232HL is on a Digi XBee Pro dev board (XBP9XR) that does not break out Port 0. There is actually not even a schematic that I can find.

closed 2021-03-01 by rkl099