rkl099/Appinventor-SerialOTG #3: micro:bit support

tguneysu, 2021-02-25:
Hi rkl099,
Thank you for a great addition to the AI2.
I am using the extension with micro:bit connected to the Android phone over an OTG & USB cables.
Below incidents are while in Companion op mode.
It works the first time AI2 Companion is started, pops up Serial Permission screen and all is fine.

On subsequent disconnect and connects, the Serial permission is never displayed, and the program does not open the USB, even though it is initialized.

Do you know why that is?

Even if I reset the APP defaults, it still does not work more than one connection attempt.

I understand this might be something in the Android, but still would like to know how to overcome it.
I am testing with various Androids, all with the same result: 
  NOTE4 & S5 with A6.0.1, NOTE@ with A4.4.2

Thanks

rkl099, 2021-02-25:
Some tips for your first problem.
Close your serial connection before you make any changes to your app  or refresh the screen.
If you need to reset the connection from the PC, remove the current instance of the Ai2 Companion in the phone. Do not just restart with the companion icon.

For Adafruit CLUE, try my debug version Appinventor-SerialOTG_Debug with support for Adafruit Feather M0, Vendor ID 239A (and some other devices). This is probably the same as used for CLUE. If not get Vendor ID and Product ID and  then I can add support for this device.
This will later be included in the release version.
Please let me know if this solves any of your problems.

tguneysu, 2021-02-26:
Hello RLK099,

I tried using your debug version and when trying to import into AI2, I got the following error:
[image]

As far as the other suggestions goes, I understand the point. But since the compiled version works OK and I only encounter it during repetitive disconnect and connects, I will manage it. So no bother.

Looking forward to testing the CLUE version , once you tell what is wrong with the extension.

Thanks, TG

rkl099, 2021-02-26:
Just downloaded and imported the extension without any problems. I'm using Linux/Firefox for the download and import to AI2, but I don't think that this matters.

tguneysu, 2021-02-26:
OK, I will retry and report. Thx.

tguneysu, 2021-02-26:
Thank you very much !
The debug version works great with Adafruit CLUE.

rkl099, 2021-02-27:
Glad to hear that it's working now!

closed 2021-03-01 by rkl099