rkl099/Appinventor-SerialOTG #4: Using the USB port while in Genymotion Emulator

tguneysu, 2021-02-28:
Hi RKL999,
I was wondering if the SerialOTG would work while I run my generated APK in the Genymotion Emulator?
It has USB support, and VirtualBox that it runs , is able to filter the PC Host USB ports.
When I run the APP it comes up fine, and the VirtualBox is set to select microBit USB connection.
However, I cannot open the port.
The reason I am asking this is, I would like to demo the AI2 software running on the PC in emulator, so I can share the display.
Would appreciate your assessment of the attempt.
Thx.

rkl099, 2021-03-01:
I have no idea!
Some general advice’s
Use an app that checks what's connected as USB OTG.
Some arm-devices needs a program that activates the serial port. Otherwise they will not be detected as a serial device. Is the Micro:bit working when connected to a real android device?
Can you cast the android screen for demonstrating purpose? But I don’t know if you can cast to a program in your pc.

tguneysu, 2021-03-05:
Just to close the loop, I wanted to let you know that I was able to use the Genymotion emulator and your SerialOTG without a problem. However, it only work with Adafruit CLUE device. None of the other devices worked, even though they worked great in the gen'ed APK. No problem though. I got my demo done in excellent quality.

closed 2021-03-05 by tguneysu

rkl099, 2021-03-18:
Glad to hear that it's working for you. It's always interesting to hear
about different applications.

ronaldbenve, 2021-07-02:
I didn't get it working with STM32 Bluepill (STM32F103), it would be nice.
VID : 0483 STMicroele
PID : 5740 STM32 Virtual Com Port
On my phone (Samsuns Galaxy S7) Bluepill works with Serial USB Terminal.