rkl099/Appinventor-SerialOTG #6: Bluetooth SPP support

JOHugo, 2021-06-13:
Hi. Would love to get Bluetooth SPP support. The serial port of my Arduino is connected to a HC-05 Bluetooth module. In windows I can pair with the HC-05 ad it looks like an normal serial port. I can even upload my HEX code from the Arduino IDE to my Arduino over Bluetooth. With AI2 I can use the Bluetooth blocks to access my Arduino, but it would be nice to integrate it with the HEX upload feature in your addon. Will this be a possibility for future releases?

rkl099, 2021-06-17:
It sounds interesting to be able to use Bluetooth for serial communication (and HEX file upload). There is support for this in the Physicaloid library, but I have not tested it yet. I will order a HC05 and do some testing to see if I can make it work with AI2.

JOHugo, 2021-06-17:
Up to now I've been using this cheaper HC-05 compatible Bluetooth module. It works fine with both 3.3V and 5V. 
https://www.aliexpress.com/item/33012208387.html
Here is the datasheet for the module. Their AT commands are also easier to use than the original HC-05
http://myosuploads3.banggood.com/products/20190129/20190129043725SKUA87502.pdf
This article explains how to connect the module + get auto reset on the DTR pin of an Arduino PRO mini working
https://forum.arduino.cc/t/bluetooth-upload-to-pro-mini-working/336021
Here is a nice video showing how to change the baud rate and use the module
https://www.youtube.com/watch?v=HxjE9l9xcSs

In my case I use Optiboot that defaults to 115200 Baud. The Bluetooth module defaults to 9600 Baud. The only thing you need to do is to change the Baud rate to 115200. Connect to the module at 9600 and do the following.
AT
AT+BAUD8    (use BAUD7 for 57600 Baud)
AT+RESET
From now on it will do serial passthrough at 115200 Baud. 
Pair it with windows. Get the serial port number assigned by Windows. Set you Arduino IDE com port and you can upload your code (Programmer type = ArduinoISP). When Windows connects to the module, the DTR line will cause the Arduino to reset and the upload will begin. Disconnect the DTR line for normal serial over Bluetooth SPP use.

rkl099, 2021-10-02:
I have examined the library that I have modified and used for my extensions. It had a Bluetooth serial class, but it only implements BT server (slave), so the external device has to initialize the connection. I think this is possible with HC-05, (not with HC-06) but this does not seems to be so useful.

JOHugo, 2021-10-04:
Thanks for the feedback and for having a look.

rkl099, 2021-10-19:
Now I have rewritten part of the library to get the server function to work.
I can see the use for reprogramming Arduino over Bluetooth, so I implemented a client function as well! . I hope that you can test the reprogramming and see if its usefull.
You will find the extension on https://github.com/rkl099/Appinventor-SerialOTG_Debug

Please return with comments.

rkl099, 2021-11-04:
I have a new version of the library and the app for testing. on SerialOTG_Debug.
It has display of file error and upload progress in nr of pages.
I have tested it on an Arduino Uno with serial and BT upload of (short) programs without any problems. I have found no reason for the long upload times that you mentioned. Maybe many retries?
If you are familiar with reading log files with Logcat Reader, then maybe you can send me some logfiles.

JOHugo, 2021-11-05:
I found the problem. I'm using Optiboot https://github.com/Optiboot/optiboot
for my boot loader and it's not compatible with any of the supported
boards. I've now installed the UNO's bootloader (think it's derived from
Optiboot) and I selected UNO as my board type. Bluetooth downloads are
working perfectly now, without the long delays :-)
One comment. The upload progress does not reset when I do another hex file
upload.
Thanks for your great work. I'm very excited about it.

closed 2022-04-16 by rkl099