rkl099/Appinventor-SerialOTG #11: Impossible to connect and driver name is empty

pedro6510, 2022-03-18:
Hello, i am trying to use your SerialOTG module to connect to an USB device and connect does not work.
Initialize is ok but driver name is empty.
Connexion parameters looks ok.

If i try with your demo app it does not work too but if i try with basic USB Serial Console app from app store i have no issue.

Do you have any idea ?
Thanks for your help.

rkl099, 2022-03-18:
Initialize only creates a communication object so its normally always ok.
Open must be done to select a device and driver for communication.
What USB-device are you using?

pedro6510, 2022-03-18:
it is a specif tracker device we are developping. 
Config used with USB serial console is really the default one : baud rate:9600, Data bits:8, Parity:None, Stop Bit:1, flow control:off
I want to develop a basic android app to use the CLI in the tracker

pedro6510, 2022-03-18:
here is the code.
[LORA_SMS.zip](https://github.com/rkl099/Appinventor-SerialOTG/files/8304109/LORA_SMS.zip)

rkl099, 2022-03-18:
My guess is that your USB-Serial device is not supported- Can you test with a linux computer to se VID and PID for your device?
You can find some debug information in the Android system log.

pedro6510, 2022-03-18:
too bad. what if i try to connect with BT ? Will it be the same ?
As my device also BT advertize it could also be a solution.

rkl099, 2022-03-18:
BT is restricted to "BT Classic". You must also pair your device with your Android device before you can use it with SerialOTG.
If you can find out your VID/PID then its maybe possible to add support.

pedro6510, 2022-03-18:
USB\VID_10C4&PID_8C0F&MI_01\7&20a5267b&0&0001
Hope you will be able to support it

[Silicon Labs - USB\VID_10C4 - 
Comes with the silicon Labs Microprocessor : EFM32LG360

rkl099, 2022-03-22:
VID 10C4 indicates a Silicon Lab device. It is assumed in Physicaloid/SerialOTG to be an USB-Serial converter CP2102
I can̈́t find any referens for PID 8C0F.

The documentation for EFM32LG360 does not say anything about the serial protocol. It is probably dependent on the software that you upload. 
Compare this dmesg output on a linux system for a programmed ATtiyn85 device.

[ 3467.735306] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 3467.735311] usb 1-1: Product: Digispark Serial
[ 3467.735315] usb 1-1: Manufacturer: digistump.com
[ 3467.769361] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[ 3467.777756] usbcore: registered new interface driver cdc_acm
[ 3467.777758] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters

cdc_acm is a function of the software on the device and the protocol used for usb serial communication.

pedro6510, 2022-03-22:
hello, here is the output requested from a linux system 

8.536637] usb 1-4: new full-speed USB device number 5 using xhci_hcd
[   78.686832] usb 1-4: New USB device found, idVendor=10c4, idProduct=8c0f, bcdDevice= 0.00
[   78.686839] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   78.686844] usb 1-4: Product: Microtracker
[   78.686848] usb 1-4: Manufacturer: Abeeway
[   78.686852] usb 1-4: SerialNumber: 2471e1035dad6b15
[   78.726200] cdc_acm 1-4:1.1: ttyACM0: USB ACM device
[   78.726425] usbcore: registered new interface driver cdc_acm
[   78.726426] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters

Regards

rkl099, 2022-03-22:
The code for selecting drivers is unnecessary complicated, still it only select drivers by VID (from the original Physicaloid Library).
I will change the code in the future, but I have tried to make a temporary solution by excluding PID 8c0f from the selection of the CP210x driver. The code should then fall back to test if its a cdc-acm device. I have no way to test it, so please test the app and extension on my Appinventor-Debug directory on github.

pedro6510, 2022-03-23:
works perfectly now thank you very much !

closed 2022-03-27 by pedro6510