rkl099/Appinventor-SerialOTG #13: Stop bits configuration

Felan412, 2022-03-29:
Requires 38400 8N2 configuration. How can I make two stop bits?

rkl099, 2022-03-30:
This is not implemented in the extension (yet), even if its included in the Physicaloid library.  Mainly because it's difficult to test.
It can be added for transmit, but it will not be tested for in the receive function.

rkl099, 2022-04-12:
I have added support for Physicaloid library stopbits in SerialOTG.
StopBits(n) and n=StopBits. Value n 0=>1, 1=>1.5, 2=>2 stopbits .
Please test and report what adapter you are using (or selected driver).
(Adapter type can be tested with a Linux system and terminal cmd dmesg) The extension can be found at github.com/rkl099/Appinventor_Debug

Felan412, 2022-04-13:
Adapter: CP2102 USB To UART Bridge Controller.
When checked by a logic analyzer, there is no difference between 1; 1.5; 2 stop bits. It is always 1 stop bit.

Felan412, 2022-04-13:
The problem seems to be that changing the stop bit before the port is open (call SerialOTG1.open) has no effect. If change the value after opening the port (for example, using a text field and a button), then everything works fine.

rkl099, 2022-04-13:
Thank you for testing and finding that problem. I have updated the extension so you can update stopbits before open, like other properties. You should also be able to change stopbits as a property in AI designer view

Felan412, 2022-04-14:
Checked, now everything works as it should. Thank you for your solution. I think that the issue can be closed.

closed 2022-04-14 by Felan412