rkl099/Appinventor-SerialOTG #15: New Error FLAG_IMMUTABLE or FLAG_MUTABLE

daniele-v, 2022-10-11:
Hello, I reopened an old application and now it gives me an error when I try to initialize the serial port. 

edu.mit.appinventor.aicompanion3: Targeting S+ (version 31 and above) requires that one of FLAG_IMMUTABLE or FLAG_MUTABLE be specified when creating a PendingIntent. Strongly consider using FLAG_IMMUTABLE, only use FLAG_MUTABLE if some functionality depends on the PendingIntent being mutable, e.g. if it needs to be used with inline replies or bubbles.
Note: You will not see another error reported for 5 seconds.

(Android 12)

Any Idea... Thanks

rkl099, 2022-10-11:
This is depending on changes in Android12. Test the updated extension in the Appinventor_Debug directory. I dont have an Android12 device, so please tell me if it works for you.

rkl099, 2022-10-17:
Updated version in Debug directory. Solves error when no USB device is connected.

closed 2024-01-19 by rkl099