rkl099/Appinventor-SerialOTG #16: ESP32C3 Support?

specternecter, 2023-01-26:
Are there any plans to support the ESP32C3 or could it be supported? As a $2 arduino compatible microcontroller with built-in USB, WiFi and BLE, it's on track to become incredibly popular in the arduino and DIY electronics world very soon. I tried to use it with this extension though and it  isn't supported. I'm just wondering if it could be.

rkl099, 2023-01-26:
Can you check   the VID and PID for the device? I assume it uses CDC as the communication protocol

specternecter, 2023-01-26:
Yes, it's CDC and it's VID is 303a and PID is 1001. For some reason they
all show up in Arduino as ESP32S3s but I don't think that really matters.
It's been that way since they came out over a year ago. But with digikey
recently hiking the price of CP2102 chips alone from about $3 to now $6.50,
I just don't see any way the C3 isn't going to take over at an even faster
pace than it already is.

specternecter, 2023-01-27:
I was curious as to how the C3 USB works and I ran across information that says the C3 isn't truly OTG capable. The S2 is OTG capable and the same price if that matters.

rkl099, 2023-01-27:
I'm just in the process of testing 1.5 beta release of SerialOTG. You can test it from SeialOTG_Debug directory
com.SerialOTG.aix and SerialOTG.aia, SerialOTG.apk. 
It now defaults to CDC-driver if no VID/PID is found. (assume that most new devices use this)
If this does not wok, you can use the force diver to CDC in the test app. This is available as an SerialOTG property in the editor or as a program function. Must be set before open!
Try this an let me know the result

specternecter, 2023-01-27:
[image]

specternecter, 2023-01-27:
I see where there's a variable for DriverName but I don't see a function to set it. Maybe I'm doing something wrong

rkl099, 2023-01-27:
Did you get it from rkl099/Appinventor_Debug directory? The problem with pending intent should be solved there.
The property and function name to force driver is USBdriver.

specternecter, 2023-01-27:
It works! You hsve to press open twice before it actually opens but I'm sure I can work that out. On other serial monitors though, you have to reset by quickly closing and re-opening the port because a hard reset also resets the port and nothing will print afterward. With this, that doesn't work and I have to close the port, do a hard reset and then when I open the port again it actually does reset again. It appears that it only forces a reset on the first opening of the port after a power up or hard reset. These are minor issues because it does work, but I'm just giving feedback. I'm amazed honestly.

rkl099, 2023-01-27:
I have some questions:
You have to press open twice. Is this when it ask for USB permission the first time?  (This is inherited from the original physicaloid library. This is something I would like to solve.)

Can you open the port without forcing CDC driver, that is using the default when VID/PID is not found?

About the reset. This could be a function of RTS/DTR status when opening the port. default is 0 for both. I think the serial terminal has 1/1 as default

specternecter, 2023-01-27:
When you press open the first time it asks for permission. Then after pressing ok, you have to press open again.

Yes, it works at default values, without forcing the driver.

For RTS/DTR, I've tried all combinations with no success.

rkl099, 2023-01-27:
Thank you, then I know. (I was to quick with RTS/DTR advice, since there are no such signals in the CDC USB interface. It should not have any impact just like changing baudrate)

specternecter, 2023-01-30:
I figured out the reset problem. When you close the port, it never actually closes. It took me a while of digging through the ESP32C3's source code to figure out how to show the port state, but I'm absolutely certain that it never closes, no-matter how many times you hit the button or what the rts/dtr is set to.

specternecter, 2023-01-31:
You can get around this issue though by informing the device that you're closing the port before you do it. Then have the device do a watchdog reset when it receives the notification that the port is closing. It's dirty, but it works.

rkl099, 2023-01-31:
I have examined the library:
Physicaloid uses UsbDeviceConnection.close() that should release all resources and close the connection.
Serial terminal/usb-serial-for-android uses UsbInterface.releaseInterface(nr)
Some examples and document suggest to use both to close the connection
I can't tell what is the correct way and I don't have any hardware to test for the moment, so you may have to live with your get around for a while.

specternecter, 2023-01-31:
Oh it doesn't bother me at all. I'm just really happy it works with the C3 now. The workaround isn't ideal as a permanent solution for the average user, but I'm completely fine with it. I'd offer to help, but I don't know how to compile an extension.

specternecter, 2023-02-05:
I found a great workaround for soft resetting the C3 and I thought I would share it in case anybody runs across this. If you add the function `((void(*)(void))0)();` to your Arduino sketch and have a reset button in the app trigger that function in the C3, it will start the sketch from the beginning but keep the port open.

closed 2024-01-19 by rkl099