rkl099/Appinventor-SerialOTG #17: Issues with Serial Wifi connection

roka80, 2023-06-22:
Hi

I am using your serialOTG.aix in my APP, and it works pretty well.
The setup consists of a Tablet (Client) connected to a Wifi Gateway (Server), which provides the serial tunnel for a single serial device.

Now, I have to change the architecture of the Hardware; instead of connecting the Tablet to the gateway hotspot, I have to connect the Gateway to the Tablet hotspot because of a multi-serial device solution.

With the new constellation, I can not Initialize the Wifi anymore. Do I have to apply a different Initialisation method since we act as a TCP server instead of a TCP client?
[image]

Have you got any idea what could be wrong?

Best regards,
Roger

rkl099, 2023-06-23:
As a server, leave net blank "". SerialOTG will use the phone/tablet TCP address and use the server mode. (See documentation SerialOTG_1.5.pdf.
Note: Only one connection is implemented.

roka80, 2023-06-23:
net blank or also dport?
What you mean with only one connection, you mean one Client only?

rkl099, 2023-06-23:
You need to set dport for the connection from the outside.
One connection: Yes, you can only make one connection from the outside.

roka80, 2023-06-23:
Ok, so you mean the outside device, which one the serial port or the TCP/IP port?
I do not need multidevices per App, I will use different APPs connected to different devices, but one App serves one device only.

rkl099, 2023-06-23:
Its the TCP/IP port. SerialOTG implement a single byte stream that can be treated in the same way (except for timing) for serial over USB, BT (classic) and TCP/IP. 
Some methods are only used for serial with an USB-Serial adapter (baud rate, parity, dtr and rt), and has no meaning for TCP/IP or BT. If you are using a serial tunnel, then you must set it up locally or include some kind of control in the byte stream.

roka80, 2023-06-24:
The strange thing is, when I use SerialOTG with Wlan adapter (on a tablet), then it works perfectly, since I use it in Hotspot mode, I can not establish a connection? Any idea why there is a difference?

rkl099, 2023-06-24:
I'm not sure I understand your problem. 
When you open SerialOTG  after InitializeWIFI with Net="" and your specified port, SerialOTG starts a background process and waits for a connection on that port and your device IP-address. After a connection is made from the outside, it behaves in the same manner as USB-Serial and BT.
Then you can read the byte stream. If no connection is made, you will have a 0 byte to read returned.
You can also use IsConnected() to test if a connection is made, but it will not always report if it is disconnected, due to the nature of the TCP/IP protocol.

closed 2024-01-19 by rkl099