rkl099/Appinventor-SerialOTG #19: Slow Data Transmission with FTDI USB to Serial Adapter in SerialOTG V1.5

hdtran2004, 2023-11-13:
Hi @rkl099,
Firstly, I want to express my gratitude for your work on this extension; it's been instrumental in the development of my application.

**Issue Description:**
I'm encountering a performance issue in my application where data transmission to an end device is significantly slower than expected. 
When sending a large text file (6100 lines, ~40 characters per line), the transmission is unusually slow.
It took about 20 minutes for the entire file to be sent.
I've observed a consistent delay of approximately 3ms between each byte of data, as shown in the attached capture.

**Setup:**
- FTDI USB to serial adaptor
- Communication settings: 38400bps, 1-Stop bit
- Test device: Pixel 7a running Android 14
- SerialOTG V1.5

**Expected Behavior:**
The data should transmit at a regular pace without such delays, especially given the baud rate and setup.

**Actual Behavior:**
Experiencing significant delays between each byte, leading to an overall slow transmission.

**Attachments:**
Transmission waveform
[image]

Could this issue be related to the driver? Any assistance or guidance you can provide would be greatly appreciated.

rkl099, 2023-11-15:
Do your example with WriteBytes (MIT forum) make  a list of length>1 ?
I looked at the code and all bytes from the list up do the device max length should be in one USB transfer to the USB-Serial converter.

hdtran2004, 2023-11-16:
I can confirm that the list length is 15 bytes (For testing I'm sending a static string), I modified the send block to log the length to confirm as per below,
[image]

rkl099, 2023-11-16:
For some reason is the write_size set to 1 in the driver (physicaloid) for  FTDI. So what you see is a function of the driver and even if you use lists will it result i bulk transfers of just one byte! I need to study the spec for different FTDI chips (and study some other drivers like Linux) to change this and also do some testing. It will take some time, but I will let you know the result.

hdtran2004, 2023-11-20:
Thank you for the confirmation, greatly appreciate your time.
Unfortunately, I'm stuck with FTDI.

closed 2024-01-19 by rkl099