rkl099/Appinventor-SerialOTG #20: RX buffer flush

Diolum, 2024-02-27:
Hi,
I seem to have a problem that I can't find a solution to.

i have a simple GUI with two button. 
One send a command and take the response and display it
and the second one does the same.

When I push either button it does not display anything ( I can see in my serial monitor the response)
If i push it a second time it works.

Restart the app:

if I push button one it does not display anything. 
if I push button two it displays the data from the button 1 push 
if I push a second time button two it display the correct data.
it seems like the RX buffer is a button press late.

if I put flush anywhere (at the start or end) it does not works anymore or I have only part or the precedent RX buffer.

here is the block view.:
[image]

Thanks for your help.

rkl099, 2024-02-28:
It seems that you expect an immediate answer  to your write of data, but it takes some time to send it over driver-USB-serial to your serial device, process the command and send the response serial-USB-driver-your program. You have to wait for the response, or you will get the response from the previous command, as you describe. (see the documentation "Communication methods" and the test programs). Use a timer to wait until you get the response.

Diolum, 2024-02-28:
Hi, yes of course !!!!!!
I have fixed the problem with a simple Wait() extension now everything is perfect. Might need to tune the delays but everything works. it it not the prettiest but enough for my use case.
thanks your extension is far better than the included one 👍
[image]

rkl099, 2024-02-28:
Good luck

closed 2024-02-28 by Diolum