rkl099/Appinventor-SerialOTG #21: BT upload - Getting .hex file from different source

mister0wl, 2024-06-02:
Hello,
at first, thank you for your work. I'd like to use SerialOTG_BT_WIFI.aia for uploading to my Nano 328pb + HC06 setup. However, I would like to download the .hex file from the Internet. The .Upload() method only accepts a board type + filepath. Would it be possible to add a method to accept board type + the content of a .hex file? 

Also, I know that the Atmel328pb uses different signature bytes (0x1e 0x95 0x16) for identification, I don't know if it matters during the upload as I don't see your code (of course I compile the Arduino code for the PB board so my hex file is OK).

Thank you,

rkl099, 2024-06-04:
Do you mean including the contents of the .hex as text after the upload command? Then it is better to download or copy it from the internet (and convert it to hex data in a programing editor if necessary)  before upload.
I don't think the signature byte is significant here. Board type is used to identify protocol (STK500 or STK500V2), memory size and baudrate. The programming of the chip is handled by the Arduino bootloader.

mister0wl, 2024-06-05:
Hello,
yes, I thought instead of passing the file path, I could pass the hex data as text. In my project I don't have control over the Android devices, and there is a demand that the update should not be written to the Android device, for security reasons. I have control over a server where I can convert the .hex file to any format (but I chose Intel .hex format as it's very easy to read and send for 8 bit microcontrollers). For my purpose, your .upload() method is monolithic. I could do the conversion and protocol handling myself, but then the benefit of this extension is lost. At first I tried to implement it using the built-in Bluetooth client component but the scheduler of the Clock component is too loose for the STK500v2 protocol and the upload failed. That's why I'd like to try my luck with your nice extension. :-) If that's not possible, I see no other option but to write an extension by myself and I'd like to save some work if it's an easy change for you to implement this new method.

rkl099, 2024-06-07:
This is a part 0of the original Physicaloid library.  I made some updates in the code to make it work 2 years ago. It seems to be to much work to implement the requested change in the library and the extension.
Ps: can you access the hex-file over the network or on a USB or SD card?

mister0wl, 2024-06-07:
OK, now I  see. I can access the file over the network; I'll download it to some shared directory, pass it to the upload() method then delete the file from the device. I hope that'll be OK for security. Thank you for your answers!

closed 2024-06-07 by mister0wl